PhilippMundhenk/BrotherScannerDocker #1: Scan doesn't work.

Bloodpilot, 2022-04-07:
Not sure if I've done something wrong but when I try to scan something the printer shows PC Connection on its display but nothing is happening.
I'm using a MFC-L2710DN.

PhilippMundhenk, 2022-04-10:
Sound like the button assignment is working, if you can trigger the scan from there, but something is wrong with the actual scan. To debug, try opening a commandline to the container and execute one of the scripts manually (e.g., scripts/scantoemail-0.2.4-1.sh). The output might give us a better indication on what's wrong.

Bloodpilot, 2022-04-10:
Thanks for your reply! 
I've actually fixed it last night, there where some permission issues when executing the start script.

closed 2022-04-10 by Bloodpilot