PhilippMundhenk/BrotherScannerDocker #2: Permission problems on QNAP

balli1187, 2022-06-07:
Dear Philipp,

I came accross your work and love the idea. Thanks for sharing!

I tried to run it on my QNAP NAS with a Docker-compose file but it rans into permission problems (see the log I copied from portainer):

`Allowing use of questionable username.
Adding user `NAS' ...
Adding new group `NAS' (1000) ...
Adding new user `NAS' (1000) with group `NAS' ...
Creating home directory `/home/NAS' ...
Copying files from `/etc/skel' ...
sh: 1: cannot create /etc/opt/brother/scanner/brscan4//brsanenetdevice4.cfg: Permission denied`

Unfortunately I was not able to sold it myself. I couldn't find the script/line in a script that causes the problem.
Can you help?

Regards Stephan

balli1187, 2022-06-09:
Hi,

Thanks for reply.
I added the lines in runScanner.sh and rebuild the Container but doesn't change anything.

I'm not so experienced in this stuff but is it right that there is a double slash in the dir that causes permission problems? 
It looks unfamiliar to me, that's why I'm asking.

PhilippMundhenk, 2022-06-09:
Hi @balli1187,
alright, I managed to update the drivers to the newest version, build the container, reproduce, and seemingly fix the error. I updated this repository. Please try to pull it again, rebuild, and recreate the container and see if it works now.

PhilippMundhenk, 2022-06-09:
Found an even better fix, make sure to grab the latest version

balli1187, 2022-06-13:
Thanks for the fix.
Now the container comes up ending wirh"startup successful" message.

Unfortunately I can not scan to my NAS. Tried 
Scanner -> Scan-to-PC -> File
But the printer shows "no computer found".

Any suggestions how to fix it?
Which protocols are needed to? I disabled SNMP in the initial setup cause I didn't know it will be used.

PhilippMundhenk, 2022-07-02:
My apologies, this topic has completely dropped off my radar.

Yes, I added this message to make sure all init works well. So we fixed the first error it seems. 

Not quite sure where the second one is coming from, I will try to reproduce. SNMP is definitely reqired though for the connection of the buttons, so this might already be an issue. You might want to keep this active.

balli1187, 2022-07-02:
Everything is fine, no need for excuse.

I figured it out myself: I changed the User that runs the container und that causes new permission problems. After changier the script's owner everything worked.

I have tried some things to reduce the size of the pdfs as my first scan needed 30mb for 4 sheets.
I'm now converting all pnm-files to jpeg before the pdf conversion starts.

Next step will be ocr and it would be really interested in that automatic correction that you mentioned in your blog.

PhilippMundhenk, 2022-07-16:
Excellent, I will close this ticket now.

But here are some pointers to the items you raise:
- I am not actually keeping the scanned documents, but instead run them through OCR, which works on much smaller file sizes (see https://github.com/PhilippMundhenk/TesseractOCRMicroservice). I am then only keeping those smaller files, enhanced with recognized text. But indeed, one could also just lower the scanning DPI, this would make scanning faster, as well. Take a look at the "resolution" setting in the scripts.
- I have never actually implemented the correction setting, it has just not bothered me enough. And I guess I got better at handling and placing the scanned documents, such that the tilting is minimal. I have, however, for another project been using something like this: https://pyimagesearch.com/2017/02/20/text-skew-correction-opencv-python/ (in fact this: https://pyimagesearch.com/2014/08/25/4-point-opencv-getperspective-transform-example/ but does not directly apply). You might want to have a go at this.

closed 2022-07-16 by PhilippMundhenk