PhilippMundhenk/BrotherScannerDocker #3: Newbie : how does it work ?

gittynick, 2023-05-05:
Hello

I have a synology nas with latest DSM 7.1 and a brother MFC-L7910DW. I would like to scan to my NAS so your script is a perfect fit for my needs ! Thanks for sharing :-)

Reading your instructions, i tweaked the run.sh file so to fit my config :

docker run -d -v /volume1/scanner:/scans -v script:/opt/brother/scanner/brscan-skey/script/ -e "NAME=Scanner" -e "MODEL=MFC-L2710DW" -e "IPADDRESS=192.168.1.55" --net=host brother

I then launched the script (via the "Scheduled tasks/user defined script")... but nothing happened :-/
Did i miss something obvious ?

Thanks for helping ;-)

gittynick, 2023-05-06:
A small follow-up : i managed to get around my install difficulties by using a ready-made container : https://registry.hub.docker.com/r/unverbraucht/brother-scanner

Unfortunately, I don't manage to mount my scanner directory... i tried "-v /volume1/scanner:/scans" and "-v /volume1/scanner:/root/brscan"
(FYI my docker log shows "/home/gittynick/brscan/brscan_2023-05-05-14-31-58.tif is created.")

How can i mount this brscan directory to my nas's Volume1/scanner directory ?
How can i create a PDF instead of a TIF ?

Thanks for helping ;-)

gittynick, 2023-05-08:
...I'm progressing ;-)
I mounted "-v /volume1/scanner:/home/gittynick/brscan" and can now retrieve my scans as TIFF files :-)
But still my last hurdle : how can i create a PDF file instead of a TIFF ?
Thanks for helping ;-)

PhilippMundhenk, 2023-05-08:
Hi @gittynick and sorry for the late reply, busy days. Synology's are always a bit fiddly with Docker, as many implementation details are not standard. Nevertheless, it should work, I also run it on my DS.

On your first issue: Try building it, but don't use the run script, this is more for testing. Better set it up through the GUI. I run it through Portainer. It will then automatically start, as well (might be a dedicated option in the GUI, not sure).
But good to know someone published an image out of it, I didn't even know that!

The mointing point in your second comment (-v /volume1/scanner:/scans) looks much better. The PDFs get created automatically and should be in /scans inside the container. You may want to check your permissions, maybe the user in the container does no have write permission for the mounted folder /volume1/scanner

gittynick, 2023-05-09:
Ok, thanks :thumbsup:
It's strange : 
- when i mount /home/gittynick/brscan i can get the (raw ?) tiff image but the /scans folder remains empty (i browsed via the console to check)
- when i try to directly mount /scans, the scanner gets unresponsive and does not start scanning at all (i checked that the permissions for both mounted folders and the container have read/write access - is there something in the environment variables that i'm missing ?)
Anyhow, i can live with the tiffs :wink:

PhilippMundhenk, 2023-08-03:
Sorry @gittynick , this dropped off my radar "a little".

Not quite sure where the .tiff are coming from. The scripts use scanadf to scan .pnm files, which in turn are converted to .pdf...

PhilippMundhenk, 2023-08-06:
Aaaah, is it possible you used the default scripts, rather than the ones in this repository? Until the current version, you had to specifically mount the given scripts into the container with -v.

I now included them in the container. Maybe use the newest version (you can now directly use the created container, see README, and don't need to manually build). This should resolve all your issues.

PhilippMundhenk, 2023-08-26:
Closing this for inactivity. Feel free to reopen, if issues persist.

closed 2023-08-26 by PhilippMundhenk