PhilippMundhenk/BrotherScannerDocker #9: Scanning not working with DCP-L2520DW

The-Engineer, 2023-08-26:
Hi Philipp,
first of all thanks for this amazing repo, very cool idea!
But I got some issues to get scanning to work.

My setup is a ubuntu 22.04 server docker host in my homelab using the following in docker compose:

```docker compose
brother-scanner:
    image: ghcr.io/philippmundhenk/brotherscannerdocker
    container_name: paperless-brother-scanner
    volumes:
        - /home/felix/docker-paperless/scans:/scans
    environment:
        - NAME=DCP-L2520DW
        - MODEL=DCP-L2520DW
        - IPADDRESS=10.23.91.54
        - UID=1000 # optional, for /scans permissions
        - GID=1000 # optional, for /scans permissions
        - TZ=Europe/Berlin # optional, for correct time in scanned filenames
    restart: unless-stopped
    network_mode: "host"
```

this starups ok and when I start a scan on the device the docker app gets notified right away, but the it runs into problems.
the logs look as followed:

``` logs
Allowing use of questionable username.
Adding user `NAS' ...
Adding new user `NAS' (1000) with group `NAS' ...
Creating home directory `/home/NAS' ...
Copying files from `/etc/skel' ...
startup successful
/opt/brother/scanner/brscan-skey/script/scantofile.sh: line 7: UID: readonly variable
filename: /scans/2023-08-26-16-30-09/2023-08-26-16-30-09-front-page%04d.pnm
scanimage: open of device brother4:net1;dev0 failed: Invalid argument
scanimage: setting of option --source failed (Invalid argument)
conversion process for 2023-08-26-16-30-09 is running in PID: 71
converting to PDF for 2023-08-26-16-30-09...
gm convert: Unable to open file (/scans/2023-08-26-16-30-09/2023-08-26-16-30-09-front-page*.pnm) [No such file or directory].
SSH environment variables not set, skipping inotify trigger.
cleaning up for 2023-08-26-16-30-09...
OCR environment variables not set, skipping OCR.
```

I also tried running the `scantofile.sh` inside the contailer which produces the following output in logs:

``` log 
filename: /scans/2023-08-26-16-42-37/2023-08-26-16-42-37-front-page%04d.pnm
./scantofile.sh: line 73:   122 Segmentation fault      (core dumped) scanimage -l 0 -t 0 -x 215 -y 297 --device-name "$device" --source "Automatic Document Feeder(centrally aligned)" --resolution $resolution --batch=$output_file
tail: /var/log/scanner.log: file truncated
./scantofile.sh: line 73:   126 Segmentation fault      (core dumped) scanimage -l 0 -t 0 -x 215 -y 297 --device-name "$device" --source "Automatic Document Feeder(centrally aligned)" --resolution $resolution --batch=$output_file
conversion process for 2023-08-26-16-42-37 is running in PID: 127
converting to PDF for 2023-08-26-16-42-37...
gm convert: Unable to open file (/scans/2023-08-26-16-42-37/2023-08-26-16-42-37-front-page*.pnm) [No such file or directory].
SSH environment variables not set, skipping inotify trigger.
cleaning up for 2023-08-26-16-42-37...
OCR environment variables not set, skipping OCR.
``` 

If you have time it would be nice if you can help :)
Best, Felix

PS: could it also be possible to update the base image to something more recent then ubuntu 16.04 ? ;)

The-Engineer, 2023-08-26:
addition: it works to manually trigger a scan from within the docker container and receive the file

```bash
root@docker-proxmox:/scans# scanimage --device "brother4:net1;dev0" --format=tiff > test.tiff
scanimage: rounded value of br-x from 211.9 to 211.881
scanimage: rounded value of br-y from 355.6 to 355.567
```

PhilippMundhenk, 2023-08-26:
I have so far only tested this with MFCs, not DCPs. I assume that the --source part is not supported, as the DCP only has a flatbed. I have added a few checks for this and will push in a second. A new docker image should be available within the next 5 to 10 minutes. Make sure to pull the new version and try with this. Let me know, if it helps.

The-Engineer, 2023-08-26:
Thank you very much for the fast reply, scanning now works 👍 
Just the fileconversion takes very long, about 3mins for a single DIN-A4 page?
and theres an unrecognized option

``` logs
2023-08-26T15:59:17.258088577Z startup successful
2023-08-26T16:00:32.280441484Z /opt/brother/scanner/brscan-skey/script/scantofile.sh: line 7: UID: readonly variable
2023-08-26T16:00:32.280469585Z filename: /scans/2023-08-26-18-00-31/2023-08-26-18-00-31-front-page%04d.pnm
2023-08-26T16:00:32.280477082Z grep: unrecognized option '--source'
2023-08-26T16:00:32.280495885Z Usage: grep [OPTION]... PATTERN [FILE]...
2023-08-26T16:00:32.280500969Z Try 'grep --help' for more information.
2023-08-26T16:00:37.281624222Z scanimage: rounded value of br-x from 211.9 to 211.881
2023-08-26T16:00:37.281654633Z scanimage: rounded value of br-y from 355.6 to 355.567
2023-08-26T16:00:37.281661245Z /opt/brother/scanner/brscan-skey/script/scantofile.sh: line 21: =: command not found
2023-08-26T16:00:39.284190490Z scanimage: rounded value of br-x from 215 to 211.881
2023-08-26T16:00:39.284268790Z scanimage: rounded value of br-y from 297 to 296.973
2023-08-26T16:00:39.284277758Z Scanning -1 pages, incrementing by 1, numbering from 1
2023-08-26T16:00:39.284299373Z Scanning page 1
2023-08-26T16:01:01.290318904Z Scanned page 1. (scanner status = 5)
2023-08-26T16:01:01.290344907Z Scanning page 2
2023-08-26T16:01:01.290351385Z scanimage: sane_start: Document feeder out of documents
2023-08-26T16:01:01.290356873Z conversion process for 2023-08-26-18-00-31 is running in PID: 72
2023-08-26T16:03:01.318493283Z converting to PDF for 2023-08-26-18-00-31...
2023-08-26T16:03:04.319288819Z SSH environment variables not set, skipping inotify trigger.
2023-08-26T16:03:04.319332149Z cleaning up for 2023-08-26-18-00-31...
2023-08-26T16:03:04.319340046Z OCR environment variables not set, skipping OCR.
```

The-Engineer, 2023-08-26:
ok, could be due to scan resolution? my regular single page scans I got so far were about 400kb in size, with your script I get 3.3mb. is there a flag to customize DPI?

PhilippMundhenk, 2023-08-26:
Thank you for testing. Seems there are still some minor things, but I will try to fix those soon.

Regarding scanning duration: Not quite, the conversion process is started, but it waits 2 minutes before even starting the conversion. If you "scan to email" within this time frame, the conversion process is aborted and the newly scanned file is merged as the rear page of the earlier scanned page (via "scan to file"). This allows to scan front and back pages, each with individual buttons, without having to manually assemble them later. This only works for a single page with a DCP, but with an MFC, this way, a whole stack of pages can be scanned without having a duplex scanner. See also: https://www.mundhenk.org/analog-document-digitization/

DPI: Currently not customizable via env, but you can always adapt the scripts and mount via e.g., -v "$PWD/script/:/opt/brother/scanner/brscan-skey/script/"

The-Engineer, 2023-08-26:
ok, thank you for the info and details about DPI.

scanning works great, even when I do two documents within the two min time window it does not interfer and process as expected.

Thank you very much for the quick help and support.
Viele Grüße aus Köln

closed 2023-08-26 by The-Engineer

PhilippMundhenk, 2023-08-26:
FYI: I added env for RESOLUTION, as well. Note that this is not an arbitrary number, but one that needs to be supported by your scanner. I also added some output to the logs on startup to show the supported capabilities of the device, including resolution.

The-Engineer, 2023-09-05:
Thanks a lot, the RESOLUTION env var works great.
Question about file sizes, do you any way to have some compression or similar to reduce filesize?
If I scan from the scanner it uses 300dpi and produces about 350kb for a single A4 page. If I go via your scripts its a 10mb file (about 30x the size).
Going through the graphicsmagick docu I did not find a flag to reduce filesize (quality flag is just for image formats not pdf). Is there maybe still a flag or different lib which could be used to convert the pnm files?

PhilippMundhenk, 2023-09-17:
Sorry for the late reply, had some busy weeks. Yes, I also noticed that the scanned files are rather large. It was never a problem for me, since I don't keep these, but submit them to my OCR microservice (https://github.com/PhilippMundhenk/TesseractOCRMicroservice), the output of which I store. 

I run every document through there, even if no text is included. The result are fairly small PDFs, e.g., 600kB instead of 3MB on a recent single page scan.

Feel free to open a new issue with a request for compression in this repo. I will see what I can do when I have a bit more time.