PhilippMundhenk/BrotherScannerDocker #10: Scripts work, but can't initiate scan from printer

LinuxOperator, 2023-10-18:
Hello, thank you for making this!!

I got the container up and running but **I'm experiencing an issue scanning from the printer using the "Scan to PC" function.**

I tried troubleshooting this by running /opt/brother/scanner/brscan-skey/script/scantofile.sh and this was successfully able to initiate a scan and produce a .pdf file.

My configuration:

>     environment:
>         - NAME=scandocker
>         - MODEL=MFC-L2717DW
>         - IPADDRESS=192.168.2.4
>         - UID=1000 # optional, for /scans permissions
>         - GID=1000 # optional, for /scans permissions
>         - TZ=America/New_York # optional, for correct time in scanned filenames
>     restart: unless-stopped
>     network_mode: "host"

Packet capture after clicking "Scan to PC" on printer:
[image]

Do you have any tips on what I could be doing wrong?
Thanks again!

PhilippMundhenk, 2023-10-19:
Could you provide the logging output from the docker container? This should help identify the issue.

It seems as if you haven't mounted any folder for the /scans directory. You may want to do that and check the permissions of that folder to match UID and GID.

LinuxOperator, 2023-10-19:
Thank you for your response!
Sorry I didn't include the full docker file, I do have a folder mounted to the /scans directory. 
When I use the scantofile.sh script the container starts the scan process and generates a .pdf file without any issue.

However when I use the printer to initiate a scan nothing happens on the docker container, nothing new appears in the scanner.log

Here is my log showing startup and execution of scantofile.sh:

```
> Allowing use of questionable username.
> Adding user `NAS' ...
> Adding new user `NAS' (1000) with group `NAS' ...
> Creating home directory `/home/NAS' ...
> Copying files from `/etc/skel' ...
> capabilities:
> [bjnp] create_broadcast_socket: ERROR - bind socket to local address failed - Cannot assign requested address
> [bjnp] create_broadcast_socket: ERROR - bind socket to local address failed - Cannot assign requested address
> scanimage: rounded value of br-x from 211.9 to 211.881
> scanimage: rounded value of br-y from 355.6 to 355.567
> All options specific to device `brother4:net1;dev0':
>   Mode:
>     --mode Black & White|Gray[Error Diffusion]|True Gray|24bit Color[Fast] [24bit Color[Fast]]
>         Select the scan mode
>     --resolution 100|150|200|300|400|600|1200|2400|4800|9600dpi [200]
>         Sets the resolution of the scanned image.
>     --source FlatBed|Automatic Document Feeder(left aligned)|Automatic Document Feeder(centrally aligned) [Automatic Document Feeder(left aligned)]
>         Selects the scan source (such as a document-feeder).
>     --brightness -50..50% (in steps of 1) [inactive]
>         Controls the brightness of the acquired image.
>     --contrast -50..50% (in steps of 1) [inactive]
>         Controls the contrast of the acquired image.
>   Geometry:
>     -l 0..211.9mm (in steps of 0.0999908) [0]
>         Top-left x position of scan area.
>     -t 0..355.6mm (in steps of 0.0999908) [0]
>         Top-left y position of scan area.
>     -x 0..211.9mm (in steps of 0.0999908) [211.881]
>         Width of scan-area.
>     -y 0..355.6mm (in steps of 0.0999908) [355.567]
>         Height of scan-area.
> startup successful
> filename: /scans/2023-10-18-10-47-03/2023-10-18-10-47-03-front-page%04d.pnm
> scanimage: rounded value of br-x from 215 to 211.881
> scanimage: rounded value of br-y from 297 to 296.973
> Scanning -1 pages, incrementing by 1, numbering from 1
> Scanning page 1
> Scanned page 1. (scanner status = 5)
> Scanning page 2
> scanimage: sane_start: Document feeder out of documents
> conversion process for 2023-10-18-10-47-03 is running in PID: 71
> converting to PDF for 2023-10-18-10-47-03...
> SSH environment variables not set, skipping inotify trigger.
> cleaning up for 2023-10-18-10-47-03...
> starting OCR for 2023-10-18-10-47-03...
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 
>   0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
> 100 4177k    0     0  100 4177k      0  3477k  0:00:01  0:00:01 --:--:-- 3481k
> 100 4177k    0     0  100 4177k      0  1896k  0:00:02  0:00:02 --:--:-- 1897k
> 100 4177k    0     0  100 4177k      0  1299k  0:00:03  0:00:03 --:--:-- 1300k
> 100 4177k    0     0  100 4177k      0   990k  0:00:04  0:00:04 --:--:--  991k
> 100 4177k    0     0  100 4177k      0   800k  0:00:05  0:00:05 --:--:--  800k
> 100 4177k    0     0  100 4177k      0   671k  0:00:06  0:00:06 --:--:--     0
> 100 4177k    0     0  100 4177k      0   578k  0:00:07  0:00:07 --:--:--     0
> 100 4177k    0     0  100 4177k      0   507k  0:00:08  0:00:08 --:--:--     0
> 100 4177k    0     0  100 4177k      0   452k  0:00:09  0:00:09 --:--:--     0
>  85 4884k    1 14211  100 4177k   1402   412k  0:08:36  0:00:10  0:08:26     0
> 100 4884k  100  707k  100 4177k  71454   412k  0:00:10  0:00:10 --:--:--     0
> SSH environment variables not set, skipping inotify trigger.
> FTP environment variables not set, skipping inotify trigger.
```

Please let me know if you have any other ideas!

PhilippMundhenk, 2023-10-19:
Interesting. The bjnp error at the top is something I am not receiving. I would assume this points to something wrong with the SNMP setup of the buttons. Weirdly enough you show that SNMP communication is happening though. Usually, SNMP errors lead to the server not showing up on the scanner at all, so you couldn't even trigger a scan then. Unfortunately, I don't have a setup at hand to record Wireshark traces for comparison.

The rest of the log does show a successful scan and OCR though. Was that triggered from the printer? I assume that this was your manual run of the script, as you say nothing shows up when you press the button.

One thing I notice from the Wireguard script is the differences in IP addresses. With this setup, IP routing is required between your container and your scanner. I am not sure if this will trip SNMP or something along the way. Do you have the option, at least for testing, to have both, server and scanner in the same subnet?

LinuxOperator, 2023-10-20:
Good idea, I tried putting the printer on the same network and experienced the same issue.

Do you know what port/protocol would be used for the printer to discover the docker container?
I am running this container in portainer, that is limiting the use of port 80 and a few others. 

Here is my latest startup log (not sure why the bjnp error is gone):
```
>Allowing use of questionable username.
> Adding user `NAS' ...
> Adding new user `NAS' (1000) with group `NAS' ...
> Creating home directory `/home/NAS' ...
> Copying files from `/etc/skel' ...
> capabilities:
> scanimage: rounded value of br-x from 211.9 to 211.881
> scanimage: rounded value of br-y from 355.6 to 355.567
> All options specific to device `brother4:net1;dev0':
>   Mode:
>     --mode Black & White|Gray[Error Diffusion]|True Gray|24bit Color[Fast] [24bit Color[Fast]]
>         Select the scan mode
>     --resolution 100|150|200|300|400|600|1200|2400|4800|9600dpi [200]
>         Sets the resolution of the scanned image.
>     --source FlatBed|Automatic Document Feeder(left aligned)|Automatic Document Feeder(centrally aligned) [Automatic Document Feeder(left aligned)]
>         Selects the scan source (such as a document-feeder).
>     --brightness -50..50% (in steps of 1) [inactive]
>         Controls the brightness of the acquired image.
>     --contrast -50..50% (in steps of 1) [inactive]
>         Controls the contrast of the acquired image.
>   Geometry:
>     -l 0..211.9mm (in steps of 0.0999908) [0]
>         Top-left x position of scan area.
>     -t 0..355.6mm (in steps of 0.0999908) [0]
>         Top-left y position of scan area.
>     -x 0..211.9mm (in steps of 0.0999908) [211.881]
>         Width of scan-area.
>     -y 0..355.6mm (in steps of 0.0999908) [355.567]
>         Height of scan-area.
> startup successful
```

PhilippMundhenk, 2023-10-21:
To be honest, I am not quite sure how the connection is established. Based on [this page](https://support.brother.com/g/b/sp/faqend.aspx?c=us_ot&lang=en&prod=mfcj995dw_us&faqid=faq00100548_001) it requires UDP: 54925 and TCP: 54921. I quick netstat confirms that. 

I believe brscan-skey configures the printer via SNMP and then waits for a connection from the printer on the open port(s), telling it which of the scan buttons (file, email, etc) the user has selected to call the according script. From there, scanadf takes over, but this part seems to work for you. 

Unless you have anything else running on this port, that shouldn't be an issue. With the error no longer being there, I would expect the port is opened accordingly.

What is the underlying setup? Is your container running natively on an OS or do you have other virtualization layers like Proxmox inbetween?

LinuxOperator, 2023-10-21:
My portainer instance is an ubuntu VM in ESXi. I just tried using docker in windows and reproduced the exact same issue. The pcap shows the same SNMP handshake then nothing.

Do I need to somehow configure the printer to communicate with the "PC" (docker)?

PhilippMundhenk, 2023-10-21:
There should be no additional configuration necessary. With the config passed via environment the brother tools configure the scanner. Also, if you don't have any configuration, you shouldn't be able to start the scanning process at all.

How are you starting this process from scanner? For me, I need to go to scan mode, then with arrows select "Scan: PC", OK, select "File", OK, Start. This triggers scantofile script in container.

I was planning on building a web interface to control scanning from smartphone rather than scanner, as I find using the buttons and display on the scanner quite tedious. Maybe the time has come. ;) Unfortunately, time is short these days...

LinuxOperator, 2023-10-25:
Yes, these are the steps I take to initiate a scan from the printer:
-Press Scan
-Press down arrow to highlight "Scan to PC"
-Press OK (selecting "Scan to PC")
-Press OK (selecting "File")
-Press Start

Following that the printer shows "Connecting to PC" for ~2minutes and gives up.
I'm thinking about installing the brother software on a PC and doing a PCAP to see how it should look....

Would it be possible to make a simple webhook that executes the scantofile.sh? For my use case I'd need that web port to be customizable. I'm happy to help with this, but I've never worked on containers before.

D0N2k, 2023-10-25:
I think I have the same issue as @LinuxOperator . When I want to start scanning on my L2710DW, nothing happens until it gets canceled. But I can initiate the scan via the command line.

I performed a packet trace through my Unifi AP and believe I have identified a possible problem. The container is registering itself with an IP from a Docker network, 172.28.0.1, while the host network is 10.10.0.0. Could it be related to this?

Here is a screenshot of the trace and my Docker Compose file.

```yaml
version: '3'

services:
  scanner:
    image: ghcr.io/philippmundhenk/brotherscannerdocker
    restart: unless-stopped
    environment:
      - NAME=Scanner
      - MODEL=MFC-L2710DW
      - IPADDRESS=10.10.0.53
      - UID=99 # optional, for /scans permissions
      - GID=100 # optional, for /scans permissions
      - TZ=Europe/Berlin # optional, for correct time in scanned filenames
    volumes:
      - /mnt/user/Dokumente/Neue Dokumente/:/scans
      - ./script:/opt/brother/scanner/brscan-skey/script/
    network_mode: "host"
```
[image]

I'm running my container on my unraid server.

PhilippMundhenk, 2023-10-25:
@LinuxOperator: Adding a simple web hook is fairly straight, I can take a look on the weekend.

@D0N2k: Interesting! That would indeed be a reason why this doesn't work. The scanner will try to contact this IP and of course fail. @LinuxOperator: Does your pcap show something similar?

Now, why does this happen? I'm not sure. The setting "network_mode=host" should make sure that your container is directly in your network and not NATed via Docker. It is possible that the Brother tooling is picking up the wrong interface for some reason. But before we check that, lets see if you have an interface available with the right IP at all: Can you open a console to the container, e.g., through Portainer or commandline and run e.g. "ip addr show | grep <you IP>" to see if this exists at all.

LinuxOperator, 2023-10-25:
My SNMP packets have the correct host IP:
[image]
My container/portainer IP:
[image]

----------

Oddly enough with my most recent test running docker in windows (host = 192.168.0.2) the SNMP packets show a IP that does not exist on my network (192.168.65.6):
[image]

My windows docker command:
`docker run -d -v "C:\Users\Peter\Documents\test⁠:/scans" -e NAME="Scanner2" -e MODEL="MFC-L2717DW" -e IPADDRESS="192.168.2.4" -e UID=1000 -e GID=1000 --net=host ghcr.io/philippmundhenk/brotherscannerdocker`

----
I think it's also worth noting, my printer doesn't recognize the docker container until AFTER I run scantofile.sh.

D0N2k, 2023-10-26:
I ran `ip addr show | grep 172.28` and it showed an interface for it. After a bit of digging in `docker network ls` revealed that it reports the host interface address from another docker network not the actual host network ip.

`ip addr show | grep 172.28`
[image]
`docker network ls`
[image]

PhilippMundhenk, 2023-10-28:
hm...I am really running out of ideas here... Not even sure if it is the same issue.

----
It seems that for @D0N2k, the Brother tooling picks up the wrong IP address. Not sure what can be done about this, or why this happens in the first place. Docker should use the host interface as outgoing interface, so this shouldn't be an issue at all.
@D0N2k: Above, you are looking for the Docker addresses. try to find an interface starting with 10.10.0, which is your host network.
----

For @LinuxOperator on the other hand, the configuration looks all fine. I have no idea why the call does not arrive at the script.

I am not quite sure regarding Windows, so I will leave that out of my considerations. I haven't worked with Docker on Windows for a while, but remember that a few years ago, I had to jump through a lot of hoops to get identical behavior to Linux systems.

Note quite sure what you mean with 

> I think it's also worth noting, my printer doesn't recognize the docker container until AFTER I run scantofile.sh.

----

One thing I did today to make it at least somewhat usable for you is integrate an minimal API and a web GUI. I documented this in the README. You can activate it, set port and even rename and disable functions via environment variables:
```
- WEBSERVER=true
- PORT=33355
- DISABLE_GUI_SCANTOIMAGE=true
- DISABLE_GUI_SCANTOOCR=true
- RENAME_GUI_SCANTOFILE="Scan front pages"
- RENAME_GUI_SCANTOEMAIL="Scan rear pages"
```
Note that the web interface is currently non-blocking. You will, however need to wait until your scan is completed before pressing another putting. I am not very good with web stuff and need to learn a bit of javascript before properly blocking the web page while the current scan is ongoing.

D0N2k, 2023-10-29:
I've tried different things now and found a way to instruct brscan-skey on the interface to use.

There is a variable in the `brscan-skey.config` called `eth`, where I was able to input my interface `br0`. After a printer restart, I was able to use the scan buttons on the device.

Additionally, I also attempted to assign a unique IP to the Docker container via ipvlan, but this resulted in the container registering with the port only and no IP.

[image]

PhilippMundhenk, 2023-10-29:
Interesting! How did you find this? I can maybe automate this setting, but this does not seem to be contained in my brscan-skey.config. Which is very weird, since we should be using the same container. Could you post the content of your brscan-skey.config (redact password, if contained)?

[image]

Aside from that, it is strange that your interface is br0, mine is ens18, but this might be related to the host system, which is Arch Linux in my case. We should be able to handle this in an automation, though.

D0N2k, 2023-10-29:
I found the parameter in the following askubuntu thread:
https://askubuntu.com/a/1448973

Other forum posts seem to suggest that brscan tries to use eth0 as default.

I'm using unraid with a bonded interface maybe that's a problem for brscan.

[image]

I hope this helps.
I'm not very proficient in Bash, or else I would have created a pull request.

PhilippMundhenk, 2023-10-29:
If this works, you can use the new version of the container. This setting is not automated and integrated into the startup of the container. Thus, brscan-skey should be configured correctly automatically.

@LinuxOperator : Not quite sure, but this might also help with your issues, maybe try out the new version. Should definitely be more stable in most cases.

D0N2k, 2023-10-29:
The commands above return the correct interface.

[image]

and its correctly set within the brscan-skey.config 

[image]

Thanks a lot. I cant speak for @LinuxOperator but my issue seems to be resolved.  👍

PhilippMundhenk, 2023-10-30:
@D0N2k: Would it be possibel to create a tcpdump from a successful communication (set up and scan)? I'd like to take a look at the protocol being used to handle the shortcut keys to see if I can get rid of the host network requirement...

D0N2k, 2023-10-31:
Sure,

Here you go:
[image]

I restarted the Docker Container at the 3rd entry and started the scan from my printer.

If you need any Information let me know.

LinuxOperator, 2023-10-31:
The new webserver/api addition is beautiful! Thanks for adding that!!
Using the printer buttons is still not working but I'm happy using the API instead.

Few side notes:
I tested changing my config from using network=host to mapping ports and everything seems to work the same:

```
 ports:
        - 54925:54925
        - 54921:54921
        - 161:161/udp
```

Additionally, not sure if I broke something or if this is a new bug. When using scantoocr.sh I'm getting an error prior to the scan being initiated. Do you have any idea what could be causing this?

```
./scantoocr.sh: line 44:   104 Segmentation fault      (core dumped) scanadf --device-name "$device" --source "Automatic Document Feeder(centrally aligned)" --resolution $resolution -o $output_file
./scantoocr.sh: line 44:   109 Segmentation fault      (core dumped) scanadf --device-name "$device" --source "Automatic Document Feeder(centrally aligned)" --resolution $resolution -o $output_file
gm convert: Unable to open file (/scans/2023-10-31-09-58-29-page*.pnm) [No such file or directory].
rm: cannot remove '/scans/2023-10-31-09-58-29-page*.pnm': No such file or directory
```

PhilippMundhenk, 2023-11-02:
Alright....so I experimented a bit more, reverse engineered the protocols a little (thank you @D0N2k!) and also build on [someone elses](https://blog.jackgaino.com/adventures-in-a-paperless-world-part-1) reverse engineering of the binary. I think I have something that may work for you now.

@LinuxOperator: Try using the newest container, you no longer need network_mode:"host". Instead, set the ports to something similar to what you have already discovered yourself:
```
ports:
        - 54925:54925/udp
        - 54921:54921
        - 161:161/udp
        - 33355:80 #optional, for webserver
```
but additionally, you will need to define the IP address under which the scanner can find your machine. I assume this is what has been failing so far. There is an undocumented option that we can use for this, however. 

If your laptop and your scanner are on the same network, this will be the address you are using to reach the web interface, as well. Depending on your setup, this may be the address of you Docker host, or your VM, etc. You might need additional settings on each virtualization layer. But this is no different than configuring access to the webserver, if this works, the scanner should be able to use the same setup to trigger a scan.

Let me know how it goes!
If this works, I will adapt the documentation accordingly.

On the scantoocr.sh: I should probably add a warning or error here. I have not maintained these scripts at all in recent years. I am only using scantofile.sh for front pages and scantoemail.sh for rear pages. The others are free to use any shell script you like. Of course you can also overwrite my scripts, by mounting new ones into the container. I will remove the broken functionality and add a warning with one of the next commits.

LinuxOperator, 2023-11-02:
Wow I'm very impressed! Just tested and I can now scan from the printer buttons. Tested using port mapping.

I'll try to mess around with OCR again this week. If I remember correctly when I first started using this the scantofile.sh would initiate OCR and it would work correctly.

Thanks for all your hard work on this!!

PhilippMundhenk, 2023-11-02:
Excellent! Glad I could help. I just love figuring out networking issues ;)

It is possible that the OCR script worked. I had initially built something there, but it was never maintained, so I decided to take it out now. The default scantofile can also do OCR and that is in fact what I have been using for more than 5 years now myself. I documented the functionality a little better in the readme: Basically, use scantofile for front pages and scantoemail for rear pages. They are sorted and interleaved into a single PDF automatically and pushed to this OCR service: https://github.com/PhilippMundhenk/TesseractOCRMicroservice
Not the fastest process when scanning front pages only (2min timeout), but works very well, if you automatically have scans stored away.

If you really want to add the OCR function again, maybe take a cue from the scantofile, switch from scanadf to scanimage and especially make sure to wrap all the arguments in quotations. This might be part of the issue above.

If you have more issues, let me know any time. I will close this issue now.

closed 2023-11-02 by PhilippMundhenk