PhilippMundhenk/BrotherScannerDocker #12: Any way for this to work in docker desktop for m1 Mac?

fivestones, 2023-10-30:
On docker desktop for Apple Silicon macs (m1, m2) the network_mode: "host" directive doesn't do anything.
I tried manually opening all the ports using 
```
    ports:
      - 33355:33355
      - 54925:54925/udp
      - 54921:54921
      - 54925:54925
```
in my docker-compose.yml, and I also tried 
```
    extra_hosts:
      - "host.docker.internal:host-gateway"
```
This works when using the scrips after I exec into the docker container to pull a scan from the scanner, but the scanner never can find the server when I try to do scan to file. Is there some other port I should open or anything else I might be able to do to make it work?

PhilippMundhenk, 2023-10-30:
Could you post the output of the container? I would expect some errors there as the installed Brother drivers for the shortcut keys to work are an amd64 package and not meant for arm. Not sure if Apple hacked some emulation layers into there, but even then I am not sure if this will work. It seems the implementation is very "non-standard", if basic things like host network mode are not supported...

I have little hope here to be honest. But lets start with the output from the container...

fivestones, 2023-10-30:
Ok, here are my logs:
```
docker-compose logs -tf
BrotherScannerToPaperless  | 2023-10-30T17:17:20.967715423Z Allowing use of questionable username.
BrotherScannerToPaperless  | 2023-10-30T17:17:20.967753923Z Adding user `NAS' ...
BrotherScannerToPaperless  | 2023-10-30T17:17:20.967756632Z Adding new user `NAS' (1000) with group `NAS' ...
BrotherScannerToPaperless  | 2023-10-30T17:17:21.033460507Z Creating home directory `/home/NAS' ...
BrotherScannerToPaperless  | 2023-10-30T17:17:21.052796090Z Copying files from `/etc/skel' ...
BrotherScannerToPaperless  | 2023-10-30T17:17:21.386343465Z using interface:
BrotherScannerToPaperless  | 2023-10-30T17:17:21.661552507Z starting webserver for API & GUI...
BrotherScannerToPaperless  | 2023-10-30T17:17:21.679570840Z running on port 33355
BrotherScannerToPaperless  | 2023-10-30T17:17:22.285907882Z webserver started
BrotherScannerToPaperless  | 2023-10-30T17:17:22.286050216Z capabilities:
BrotherScannerToPaperless  | 2023-10-30T17:17:22.306705257Z Output format is not set, using pnm as a default.
BrotherScannerToPaperless  | 2023-10-30T17:17:22.969544049Z Created directory: /var/lib/snmp/cert_indexes
BrotherScannerToPaperless  | 2023-10-30T17:17:27.369999385Z scanimage: rounded value of br-x from 211.9 to 211.881
BrotherScannerToPaperless  | 2023-10-30T17:17:27.370101427Z scanimage: rounded value of br-y from 355.6 to 355.567
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404738135Z
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404799802Z All options specific to device `brother4:net1;dev0':
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404803927Z   Mode:
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404806885Z     --mode Black & White|Gray[Error Diffusion]|True Gray|24bit Color[Fast] [24bit Color[Fast]]
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404810510Z         Select the scan mode
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404814010Z     --resolution 100|150|200|300|400|600|1200|2400|4800|9600dpi [200]
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404816802Z         Sets the resolution of the scanned image.
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404819885Z     --source FlatBed|Automatic Document Feeder(left aligned)|Automatic Document Feeder(centrally aligned) [Automatic Document Feeder(left aligned)]
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404823093Z         Selects the scan source (such as a document-feeder).
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404825968Z     --brightness -50..50% (in steps of 1) [inactive]
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404828552Z         Controls the brightness of the acquired image.
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404831385Z     --contrast -50..50% (in steps of 1) [inactive]
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404834218Z         Controls the contrast of the acquired image.
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404837010Z   Geometry:
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404839760Z     -l 0..211.9mm (in steps of 0.0999908) [0]
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404842635Z         Top-left x position of scan area.
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404845427Z     -t 0..355.6mm (in steps of 0.0999908) [0]
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404848218Z         Top-left y position of scan area.
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404851010Z     -x 0..211.9mm (in steps of 0.0999908) [211.881]
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404854427Z         Width of scan-area.
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404857218Z     -y 0..355.6mm (in steps of 0.0999908) [355.567]
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404860010Z         Height of scan-area.
BrotherScannerToPaperless  | 2023-10-30T17:17:27.404884802Z
BrotherScannerToPaperless  | 2023-10-30T17:17:27.430291843Z startup successful
```

As for amd64 vs arm64, docker desktop uses qemu virtualization on macos as far as I understand and so I haven't had any trouble running a bunch of different docker images that weren't complied for arm64. I do get a message on this one which I haven't seen before when I do docker-compose up -d: 
```
docker-compose up -d
[+] Building 0.0s (0/0)                                                                                                                                                                                                                                               docker:desktop-linux
[+] Running 3/3
 ✔ Container brotherscaner-ocr-1                                                                                                                                  Running                                                                                                             0.0s
 ✔ Container BrotherScannerToPaperless                                                                                                                            Started                                                                                                            10.2s
 ! brother-scanner The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
```
But if I add `    platform: linux/amd64` in my docker-compose.yml file, that message goes away:
```
docker-compose up -d
[... cut ...]                                   0.0s
 ✔ Container BrotherScannerToPaperless  Started                                                                                                                                                                                                                                      10.3s
```
And either way, I don't see any difference in the logs or in how the app works (at least, I am able to initiate a scan from the scripts in both cases).

Thanks for your help!

PhilippMundhenk, 2023-10-30:
It's a bit of a shot in the dark, as I can't get it to work here, but just for testing, what you could try is to launch from any computer that can reach your scanner something like this:
```
snmpset -v 1 -c internal <YourScannerIP> 1.3.6.1.4.1.2435.2.3.9.2.11.1.1.0 s 'TYPE=BR;BUTTON=SCAN;USER="TEST";FUNC=FILE;HOST=<YourHostIP>:54935;APPNUM=1;DURATION=360;BRID=;'
```
Where <YourScannerIP> is the IP address of your scanner and <YourHostIP> is the IP of the host your Docker container is running on. This should create a virtual shortcut button (i.e., on your scanner, select "Scan to PC", File, TEST, Start) which may or may not trigger the Brother driver in the container.

PhilippMundhenk, 2023-10-30:
Never mind, I set up my container without host network and tried this and while the setting of the parameters on the scanner works, it seems the trigger does not reach the server. Maybe there is some state involved in the driver, as well. Or I am missing something else entirely...

PhilippMundhenk, 2023-11-02:
Alright....so I experimented a bit more, reverse engineered the protocols a little (see also #10) and also build on [someone elses](https://blog.jackgaino.com/adventures-in-a-paperless-world-part-1) reverse engineering of the binary. I think I have something that may work for you now.

Try using the newest container, you no longer need network_mode:"host". Instead, set the ports to something similar to what you have already discovered yourself:

ports:
        - 54925:54925/udp
        - 54921:54921
        - 161:161/udp
        - 33355:80 #optional, for webserver

but additionally, you will need to define the IP address under which the scanner can find your machine. There is an undocumented option that we can use for this.

If your laptop and your scanner are on the same network, this will be the address you are using to reach the web interface, as well. Depending on your setup, this may be the address of you Docker host, or your VM, etc. You might need additional settings on each virtualization layer. But this is no different than configuring access to the webserver, if this works, the scanner should be able to use the same setup to trigger a scan.

Let me know how it goes!
If this works, I will adapt the documentation accordingly.

fivestones, 2023-11-03:
Thanks! I tried this out, still having some trouble though. I can see the "NAS" user the container is sending to the scanner on the scanner display when I do Scan to PC > file > NAS which I don't think was there before. But then it just says "connecting to computer" and times out. I did a tcpdump to see what it was doing, and there is a message coming from the scanner over udp, but it doesn't say what the message is and the container doesn't seem to respond.

tcpdump output:
```
sudo tcpdump host 192.168.1.85
tcpdump: data link type PKTAP
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on pktap, link-type PKTAP (Apple DLT_PKTAP), snapshot length 524288 bytes
19:03:30.196539 IP 192.168.210.2.51397 > brwd8126533a359.snmp:  C="internal" SetRequest(475)  E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=IMAGE;HOST=192.168.1.168:54925;APPNUM=1;DURATION=360;BRID=;" E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=OCR;HOST=192.168.1.168:54925;APPNUM=3;DURATION=360;BRID=;" E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=EMAIL;HOST=192.168.1.168:54925;APPNUM=2;DURATION=360;BRID=;" E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=FILE;HOST=192.168.1.168:54925;APPNUM=5;DURATION=360;BRID=;"
19:03:30.196697 IP kepler.34614 > brwd8126533a359.snmp:  C="internal" SetRequest(475)  E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=IMAGE;HOST=192.168.1.168:54925;APPNUM=1;DURATION=360;BRID=;" E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=OCR;HOST=192.168.1.168:54925;APPNUM=3;DURATION=360;BRID=;" E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=EMAIL;HOST=192.168.1.168:54925;APPNUM=2;DURATION=360;BRID=;" E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=FILE;HOST=192.168.1.168:54925;APPNUM=5;DURATION=360;BRID=;"
19:03:30.605301 IP brwd8126533a359.snmp > kepler.34614:  C="internal" GetResponse(475)  E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=IMAGE;HOST=192.168.1.168:54925;APPNUM=1;DURATION=360;BRID=;" E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=OCR;HOST=192.168.1.168:54925;APPNUM=3;DURATION=360;BRID=;" E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=EMAIL;HOST=192.168.1.168:54925;APPNUM=2;DURATION=360;BRID=;" E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=FILE;HOST=192.168.1.168:54925;APPNUM=5;DURATION=360;BRID=;"
19:03:30.605346 IP brwd8126533a359.snmp > 192.168.210.2.51397:  C="internal" GetResponse(475)  E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=IMAGE;HOST=192.168.1.168:54925;APPNUM=1;DURATION=360;BRID=;" E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=OCR;HOST=192.168.1.168:54925;APPNUM=3;DURATION=360;BRID=;" E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=EMAIL;HOST=192.168.1.168:54925;APPNUM=2;DURATION=360;BRID=;" E:2435.2.3.9.2.11.1.1.0="TYPE=BR;BUTTON=SCAN;USER="NAS";FUNC=FILE;HOST=192.168.1.168:54925;APPNUM=5;DURATION=360;BRID=;"
19:03:34.534768 IP kepler.56797 > brwd8126533a359.54921: Flags [S], seq 1228870531, win 65535, options [mss 1460,nop,wscale 6,nop,nop,TS val 598267194 ecr 0,sackOK,eol], length 0
19:03:34.540968 IP brwd8126533a359.54921 > kepler.56797: Flags [S.], seq 1405693976, ack 1228870532, win 8688, options [mss 1460,nop,wscale 0,nop,nop,sackOK,nop,nop,TS val 235462050 ecr 598267194], length 0
19:03:34.541188 IP kepler.56797 > brwd8126533a359.54921: Flags [.], ack 1, win 2058, options [nop,nop,TS val 598267200 ecr 235462050], length 0
19:03:34.547846 IP brwd8126533a359.54921 > kepler.56797: Flags [P.], seq 1:10, ack 1, win 8688, options [nop,nop,TS val 235462050 ecr 598267200], length 9
19:03:34.547979 IP kepler.56797 > brwd8126533a359.54921: Flags [.], ack 10, win 2058, options [nop,nop,TS val 598267207 ecr 235462050], length 0
19:03:35.675178 IP kepler.56797 > brwd8126533a359.54921: Flags [P.], seq 1:5, ack 10, win 2058, options [nop,nop,TS val 598268334 ecr 235462050], length 4
19:03:35.692453 IP brwd8126533a359.54921 > kepler.56797: Flags [P.], seq 10:85, ack 5, win 8684, options [nop,nop,TS val 235463200 ecr 598268334], length 75
19:03:35.692647 IP kepler.56797 > brwd8126533a359.54921: Flags [.], ack 85, win 2057, options [nop,nop,TS val 598268351 ecr 235463200], length 0
19:03:35.733726 IP kepler.56797 > brwd8126533a359.54921: Flags [F.], seq 5, ack 85, win 2057, options [nop,nop,TS val 598268393 ecr 235463200], length 0
19:03:35.740500 IP brwd8126533a359.54921 > kepler.56797: Flags [.], ack 6, win 8683, options [nop,nop,TS val 235463250 ecr 598268393], length 0
19:03:35.740502 IP brwd8126533a359.54921 > kepler.56797: Flags [FP.], seq 85, ack 6, win 8683, options [nop,nop,TS val 235463250 ecr 598268393], length 0
19:03:35.740729 IP kepler.56797 > brwd8126533a359.54921: Flags [.], ack 86, win 2057, options [nop,nop,TS val 598268399 ecr 235463250], length 0
19:03:44.663944 IP brwd8126533a359.40268 > kepler.54925: UDP, length 118
19:03:46.582238 IP brwd8126533a359.40268 > kepler.54925: UDP, length 118
```

and docker logs:
```
docker-compose logs -tf
brotherscaner-brother-scanner-1  | 2023-11-03T00:03:29.226088677Z setting up user & logfile:
brotherscaner-brother-scanner-1  | 2023-11-03T00:03:29.254004427Z groupadd: GID '20' already exists
brotherscaner-brother-scanner-1  | 2023-11-03T00:03:29.432531302Z Allowing use of questionable username.
brotherscaner-brother-scanner-1  | 2023-11-03T00:03:29.432552093Z Adding user `NAS' ...
[... cut ...]
01:17:52.436377 IP 889f4cad6871.57755 > 192.168.65.7.domain: 14672+ PTR? 1.0.22.172.in-addr.arpa. (41)
01:17:52.484594 IP 192.168.65.7.domain > 889f4cad6871.57755: 14672 NXDomain 0/0/0 (41)
01:17:53.355022 IP 192.168.65.1.57066 > 889f4cad6871.54925: UDP, length 118
01:17:53.376180 IP 889f4cad6871.56001 > 192.168.65.7.domain: 1791+ PTR? 1.65.168.192.in-addr.arpa. (43)
01:17:53.425613 IP 192.168.65.7.domain > 889f4cad6871.56001: 1791 NXDomain 0/0/0 (43)
01:17:55.291475 IP 192.168.65.1.57066 > 889f4cad6871.54925: UDP, length 118
```

Not sure where to go from here.

fivestones, 2023-11-03:
Maybe I could make a total hack to monitor tcpdump for a UDP message inside the container and when it comes, run the scantofile.sh script. Ha!
If I get time.
That's probably a pretty fragile solution though, and only would be able to use one of the 4 scanner button functions I guess.

fivestones, 2023-11-03:
Also now for the web gui I'm only getting a 500 service error, both if I check from inside the container and from outside. Not sure if something about that got changed in the last update.

PhilippMundhenk, 2023-11-03:
First on the webserver: There shouldn't be any issues there. I haven't changed anything recently. Maybe check out the `/var/log/lighttpd/error.log`, if there is anything there.

On the connectivity issues, I fear I need some more information. There seems to be some strange things going on:
- What I can get from the above is that your Mac IP visible towards the scanner is 192.168.1.168. From the tcpdump your printer is at 192.168.1.85. I assume a /24 subnet.
- From your first tcpdump (on the host?), I see that there is a connection to the printer established from 192.168.210.2 though. This may be the internal network of the VMs, but there is likely a mismatch to your HOST_IPADDRESS of 192.168.1.168.
- In both tcpdumps, I am also missing the connection from scanner to port 161 of your Mac. There should be an additional message to port 161/udp when scanning starts. Is this port opened (on Mac, VM, Docker)?
- The second tcpdump then also introduces a third network with 192.168.65.7, not sure where this is coming from. As well as a fourth network with 172.22.0.1, which I assume to be the Docker internal network.

So from what can gather here, there is at least one network too much. Try the following:
- Please post your complete docker-compose config.
- Record the tcpdumps to file (use -w) and drop the files here. Make sure to remove any confidential information first. The headlines above are a little hard to debug.
- Check if you can access the web interface (even with error 500) from a different device on the network of the scanner. If this is not possible, there is some misconfiguration for your VM & Docker setup. Refer to the Mac documentation on how to open ports to the VM in which Docker is running, as well as on your Mac. If it is possible, the IP you are using for this is the same as the HOST_IPADDRESS parameter for the container.

PhilippMundhenk, 2024-01-28:
Closing this now. Feel free to reopen, if more support is required

closed 2024-01-28 by PhilippMundhenk