PhilippMundhenk/BrotherScannerDocker #18: FTPS Upload fails

3lementX, 2024-04-08:
I tried to set up the FTPS upload today. I encountered two problems:

1. the FTPS upload is only triggered if an OCR server is configured. Without an OCR server, the script is not called.
2. after I have set up the OCR server, I get the following error message for the FTPS upload: 

`curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.se/docs/sslcerts.html
curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
Uploading to ftp failed while using curl`

I could not solve this problem. Is there a solution to this problem?
Thanks.

3lementX, 2024-04-15:
I found a solution with the help of ChatGPT that works for me:

`user=$1
password=$2
address=$3
filepath=$4

cd /scans

if [ -z "${user}" ] || [ -z "${password}" ] || [ -z "${address}" ] || [ -z "${filepath}" ]; then
  echo "FTP environment variables not set, skipping inotify trigger."
else
  for file in *; do
    if [ -f "$file" ]; then
      if curl --silent \
          --show-error \
          --user "${user}:${password}" \
          --upload-file "$file" \
          "ftp://${address}${filepath}" ; then
        echo "Uploading $file to ftp server ${address} successful."
        rm "$file" # Datei löschen, nachdem sie erfolgreich hochgeladen wurde
      else
        echo "Uploading $file to ftp failed while using curl"
        echo "user: ${user}"
        echo "address: ${address}"
        echo "filepath: ${filepath}"
        exit 1
      fi
    fi
  done
fi`

3lementX, 2024-04-15:
I found a solution with the help of ChatGPT that works for me:

```
user=$1
password=$2
address=$3
filepath=$4

cd /scans

if [ -z "${user}" ] || [ -z "${password}" ] || [ -z "${address}" ] || [ -z "${filepath}" ]; then
  echo "FTP environment variables not set, skipping inotify trigger."
else
  for file in *; do
    if [ -f "$file" ]; then
      if curl --silent \
          --show-error \
          --user "${user}:${password}" \
          --upload-file "$file" \
          "ftp://${address}${filepath}" ; then
        echo "Uploading $file to ftp server ${address} successful."
        rm "$file" # Datei löschen, nachdem sie erfolgreich hochgeladen wurde
      else
        echo "Uploading $file to ftp failed while using curl"
        echo "user: ${user}"
        echo "address: ${address}"
        echo "filepath: ${filepath}"
        exit 1
      fi
    fi
  done
fi
```

Thank you for your work!

PhilippMundhenk, 2024-04-28:
My apologies for the late reply, had a few busy weeks.

Yes, you solution will work, but please take note that you are now using an insecure connection. FTP by default is unencrypted, FTPS uses encryption through TLS. With your script, you have turned off the encryption. This may be acceptable, depending on your use case, but by default, we do not support this. I might add a config option. Tracking this in #19