PhilippMundhenk/BrotherScannerDocker #20: Printer always displays QNAP machine name as "NAS"

pedropombeiro, 2024-09-12:
First of all, thanks for making this great work public! I quickly found it after wondering if someone had thought of packaging the Brother tool as a Docker image, so you saved me a lot of time!

I tried customizing the name of the target machine, by setting `NAME=Paperless`, but I always see `NAS` as an option on my printer. Not sure if that is due to the fact that that is the hostname of my QNAP. Any idea?

pedropombeiro, 2024-09-12:
Indeed, I changed `USERNAME` to `Paperless`, and immediately saw a new `Paperless` entry in the Scan menu.

PhilippMundhenk, 2024-09-15:
Oh, very good catch, thank you! I never noticed, as I always run with the same parameters. The driver takes a name during setup, but never displays that. Instead, the username is displayed. Should be an easy fix though (testing aside). Will do asap.

closed 2024-10-06 by pedropombeiro