PhilippMundhenk/BrotherScannerDocker #30: Invoking of scripts from webserver is not working

pedropombeiro, 2024-09-16:
I tried enabling the webserver, and although I can see the page, clicking on the buttons returns immediately and does nothing. I tried looking at the lightttpd logs, but only the startup line is there. Any ideas on how I could get more information on what is happening?

```yaml
  brother-scanner:
    image: ghcr.io/philippmundhenk/brotherscannerdocker
    extends:
      file: ../common-services.yml
      service: baseapp
    container_name: docs_brother_scanner
    volumes:
      - ${paperlessConsumeDir}:/scans/
    ports:
      - 33355:80
      - 54925:54925/udp # Scan Key Tool UDP: mandatory, for scanner tools
      - 54921:54921 # Scan Key Tool TCP: mandatory, for scanner tools
      - 161:161/udp # SNMP: mandatory, for scanner tools
    environment:
      - NAME=Scanner
      - USERNAME=Paperless
      - MODEL=MFC-J4335DW
      - IPADDRESS=192.168.46.12
      - UID=500 # note: network mount needs to have correct permissions!
      - GID=101 # note: network mount needs to have correct permissions!
      - HOST_IPADDRESS=192.168.96.10
      - RESOLUTION=200
      - WEBSERVER=true
      - RENAME_GUI_SCANTOFILE="Scan front pages"
      - RENAME_GUI_SCANTOEMAIL="Scan rear pages"
      - DISABLE_GUI_SCANTOOCR=true
      - DISABLE_GUI_SCANTOIMAGE=true
    restart: unless-stopped
```

```shell
$ xh -vv GET 'http://nas:33355/scan.php' target==file
GET /scan.php?target=file HTTP/1.1
Accept: */*
Accept-Encoding: gzip, deflate, br
Connection: keep-alive
Host: nas:33355
User-Agent: xh/0.22.2

HTTP/1.1 200 OK
Accept-Ranges: bytes
Content-Length: 0
Content-Type: text/html; charset=UTF-8
Date: Mon, 16 Sep 2024 22:22:33 GMT
Server: lighttpd/1.4.63
```

PhilippMundhenk, 2024-09-17:
Hm... Is this the new version? We just merged a major upgrade of the UI a few hours ago.

In any case, take a look at the container logs. Most important output gets rerouted there. This might help debug the issue.

pedropombeiro, 2024-09-17:
That doesn't seem to be the case for me, the scripts are executable:

[image]

pedropombeiro, 2024-09-17:
I'll close this for now, since the functionality is not vital for me.

closed 2024-09-17 by pedropombeiro

PhilippMundhenk, 2024-09-17:
Would be interesting to see the logs though to see if something is going wrong that I haven't seen yet. Typically, issues surface again elsewhere at some point...

pedropombeiro, 2024-09-17:
From what I recall, the logs didn't contain anything as a response to the button click.

PhilippMundhenk, 2024-09-17:
That's ideed strange. Well, if you have issues again, let me know and we'll look into it!