PhilippMundhenk/BrotherScannerDocker #31: Scanning initiated by scanner buttons starts with a great delay

archont00, 2024-09-17:
Hi,

The scanner is connected via ethernet cable.

```
device `brother4:net1;dev0' is a Brother *DCP-B7520W Brother_DCP-B7520DW_series
```

When I use in `scripts/scantofile.sh`:

```
scan_cmd="scanimage -l 0 -t 0 -x 215 -y 297 --device-name=$device --resolution=$resolution --batch=$output_file"
echo "Command: $scan_cmd"
```
**the initiation of the scanning has a great delay of cca 30 seconds** ("Connecting to PC" is displayed on the scanner LCD):

```
2024-09-17T**19:32:54**.578402000Z /opt/brother/scanner/brscan-skey/script/scantofile.sh: line 14: UID: readonly variable
2024-09-17T19:32:54.579085000Z filename: /scans/2024-09-17--21-32-54/2024-09-17--21-32-54-front-page%04d.pnm
2024-09-17T19:32:54.579236000Z Command: scanimage -l 0 -t 0 -x 215 -y 297 --device-name=brother4:net1;dev0 --resolution=150 --batch=/scans/2024-09-17--21-32-54/2024-09-17--21-32-54-front-page%04d.pnm
2024-09-17T19:32:54.579382000Z Output format is not set, using pnm as a default.
2024-09-17T**19:33:03**.579674000Z scanimage: rounded value of br-x from 215 to 211.881
2024-09-17T19:33:03.580342000Z scanimage: rounded value of br-y from 297 to 296.973
2024-09-17T19:33:03.580492000Z Scanning infinity pages, incrementing by 1, numbering from 1
2024-09-17T19:33:03.580631000Z Scanning page 1
2024-09-17T**19:34:23**.590222000Z Scanned page 1. (scanner status = 5)
```
Scanning of subsequent pages within the batch starts promptly after button press (Next = up arrow key on the scanner).

When I run the above `scanimage` command from terminal inside the docker container, it takes 9 seconds to actually start the scanning. The scanning itself takes just a few seconds as it is 150 dpi only.

As per above log, the delay between the button press (19:32:54) and the scanning (cca 19:33:34 less few seconds spent on actual scanning) looks like 3 times the 9 seconds delay.

**Any ideas what might be wrong with `brother4` driver or `brscan-skey` and if it can be setup to run with shorter initial delay?** It makes it impractical to use the scanner buttons now...

BTW, I tried to run `scanimage` in terminal inside the container with `--device-name=escl:http://10.11.13.11:80` instead and the scanning started immediately without any delay.

Unfortunately, when I tried to use this device name in the script, the delay of cca 10 seconds is still there and the command fails eventually (driver is probably not supported by `brscan-skey`?):

```
2024-09-17T**19:35:35**.600366000Z /opt/brother/scanner/brscan-skey/script/scantofile.sh: line 14: UID: readonly variable
2024-09-17T19:35:35.601023000Z filename: /scans/2024-09-17--21-35-34/2024-09-17--21-35-34-front-page%04d.pnm
2024-09-17T19:35:35.601164000Z Command: scanimage -l 0 -t 0 -x 215 -y 297 --device-name=escl:http://10.11.13.11:80 --resolution=100 -v -v -v --batch=/scans/2024-09-17--21-35-34/2024-09-17--21-35-34-front-page%04d.pnm
... cut on purpose ...
2024-09-17T19:35:35.602824000Z Scanning page 1
2024-09-17T**19:35:45**.601742000Z scanimage: sane_start: Invalid argument
2024-09-17T19:35:45.602384000Z Batch terminated, 0 pages scanned
2024-09-17T19:35:45.602531000Z Closing device
2024-09-17T19:35:45.602666000Z Calling sane_exit
```

archont00, 2024-09-17:
Exactly: The delay of cca 30 seconds occurs when I press the scan button directly on the printer.

PhilippMundhenk, 2024-09-22:
My apologies for the late reply, I had a bit of a busy week.

tl;dr: No obvious solution, sorry.

Interesting...I do observer rather long delays as well, when using the button, especially when using the flatbed, rather than the auto feeder. In my case, this is 8s though, which is much more bearable. With ADF, it is closer to 2s. Thus, 30s is really an extremely long time.

I do call the scan commands twice, with some sleep padding, as we otherwise observed issues on some printers, but once scanimage is triggered, timing is outside of my sphere of influence. I have not reverse engineered the scanner driver, it is very much possible, there are issues there. I also don't want to exclude the possibility that the current configuration in this repo is not ideal for every scanner. It is posible that one could reach faster scanning times with better settings, as you did above. A few things I noticed:
- Calling with ```--device-name=$device```, as you describe above, currently might not work. Thus the default does not use the device name, yet. Once #33 is in, this should be fixed.
- The device name (and thus driver) is currently not configurable, but this would be a fairly easy addition. However, seeing that this does not help with triggering through the buttons, lets not use this as the first step.
- It is posible there are some implicit dependencies between brscan-skey and brscan4. I assume that the scanner, once triggering is performed via buttons, is in a state that does not allow it to be triggered via other means. One would need to terminate/abort the SNMP sequence manually in the script to enable scanning in this case. We had reverse engineered some of the protocol in #10, but I would need some time to read into this again. The fastest would be to just move scanning to detached process, but this would disable all further control, such as scanning additional pages.
- It might be possible to also try some of the other SANE commands, like scanadf, but I have very little experience with this. I had more failures than success with these.

Let me think about this a little more, but ad the moment, I don't quite see an easy solution to this, unfortunately...

PhilippMundhenk, 2024-09-22:
out of curiosity: Which scanner are you using? Workflows and buttons seem very different from mine.

archont00, 2024-09-23:
Hi,
I use DCP-B7520W, which has only the flatbed scanner (no ADF).
Thanks for the comments - I will try to test with other commands from SANE. Also, there is a binary program for scanning provided by Brother within the scanner driver package and its original scanning scripts. 
I will get back if I find anything.

PhilippMundhenk, 2026-10-10:
Thank you again for the detailed report and the timestamps, they made it clear where the time goes. I revisited this while going through all open issues today, and I want to be honest about what is and is not in reach.

The 30 seconds sit between the button press and the moment `scanimage` starts receiving data ("Connecting to PC" on the panel, then the first page). That handshake happens entirely inside Brother's closed driver and the device: the container only runs `scanimage` once the driver hands over the button event. So I cannot speed it up directly, and I have no DCP-B7520DW to dig into the timing on the wire.

Two things have changed since then that may be worth one try on the current image:

- The whole scan path was rewritten and the container now keeps the device's "Scan to PC" registration fresh every two minutes. On my own device the registration used to expire silently, and a button press after that first had to re-establish the connection. If part of your 30 seconds was that, it should be shorter now.
- The exact `scanimage` command and its timing are logged in `scanner.log`, so a single button scan gives a precise picture of where the time goes.

If you give it a try, the one thing that would tell us whether this is the driver or something about the setup is a comparison with Brother's own `brscan-skey` and `scanimage` run directly on a Linux machine (outside Docker) with the same device. If the delay is the same there, it is the driver and there is nothing more this project can do; if it is faster, please reopen this issue with both logs and I will dig in.

Closing it for now with that caveat. Thanks for the patience, and for the kind words earlier!

closed 2026-10-10 by PhilippMundhenk