PhilippMundhenk/BrotherScannerDocker #49: Stable docker release

Forceflow, 2024-09-25:
Hey there,

Been using this project for a while now, I think the recent flurry of activity (great to see the project moving, thanks to all contributors) has broken the docker builds. (The scan buttons on the web page suddenly stopped working for me after an update).

I understand that the project is moving forward, but maybe it would be interesting to provide a stable docker tag as well, much like you do versioned releases for the non-docker version.

All the best!

(just for completeness, here's my docker-compose)

`---
version: "3"
services:
    brother-scanner:
        image: ghcr.io/philippmundhenk/brotherscannerdocker
        container_name: brother-scanner
        ports:
            - 54925:54925/udp # mandatory, for scanner tools
            - 54921:54921 # mandatory, for scanner tools
            - 161:161/udp # mandatory, for scanner tools
            - 33355:33355
        environment:
            - NAME=Scanner
            - MODEL=DCP-L2520DW
            - IPADDRESS=192.168.1.105
            - UID=1000 # note: network mount needs to have correct permissions!
            - GID=1000 # note: network mount needs to have correct permissions!
            - TZ=Europe/Brussels
            - HOST_IPADDRESS=192.168.1.121
            - WEBSERVER=true # optional, activates GUI & API
            - PORT=33355 # optional, sets port for webserver (default: 80)
            - USE_JPEG_COMPRESSION=true
        restart: unless-stopped

PhilippMundhenk, 2024-09-25:
Hi, you are absolutely right! We went from a few changes a year, if at all, to quite some activity within days. I was not prepared for that and we hit a few bumps along the way. To avoid these bumps affecting everyone, a few days ago, I create the [first release](https://github.com/PhilippMundhenk/BrotherScannerDocker/releases/tag/v1.0.0). You may want to use ```image: ghcr.io/philippmundhenk/brotherscannerdocker:refs-tags-v1.0.0``` instead. This should give you a stable version where also the buttons should work (just tested with your configuration, seems to work for me).
If that is not the case, please let us know again!

btw: From your docker-compose, it looks like you are running the default scripts. In that case, you may also want to disable the additional buttons, and rename the default buttons. Under ```environment:``` add:
```
- DISABLE_GUI_SCANTOIMAGE=true
- DISABLE_GUI_SCANTOOCR=true
- RENAME_GUI_SCANTOFILE="Scan front pages"
- RENAME_GUI_SCANTOEMAIL="Scan rear pages"
```

PhilippMundhenk, 2024-09-25:
Oh, the new web interface makes extensive use of JS. Which browser are you using? This could be an issue.

@vgarcia007: Could we have a fallback to plain HTML buttons/interface (of course without disabling while scanning, etc.) if JS is not available in browser?

Forceflow, 2024-09-26:
> Oh, the new web interface makes extensive use of JS. Which browser are you using? This could be an issue.
> 
> @vgarcia007: Could we have a fallback to plain HTML buttons/interface (of course without disabling while scanning, etc.) if JS is not available in browser?

I was using Firefox, also tested Edge. Didn't see any output in JS console or docker log.

PhilippMundhenk, 2024-09-27:
Does it work with v1.0.0?

PhilippMundhenk, 2024-10-11:
I assume this is working. Feel free to reopen, if issue persists

closed 2024-10-11 by PhilippMundhenk