PhilippMundhenk/BrotherScannerDocker #55: Web server non-functional, no logs

pedropombeiro, 2024-10-11:
I decided to give the webserver a try, but it is not triggering scans for me. I even tried running the curl command inside the container, but still no results, and nothing logged:

[image]

Any ideas what I could look for?

Here's my docker-compose config:

```yaml
  brother-scanner:
    image: ghcr.io/philippmundhenk/brotherscannerdocker:pedropombeiro-convert-to-python3
    extends:
      file: ../common-services.yml
      service: baseapp
    container_name: docs_brother_scanner
    volumes:
      - ${paperlessConsumeDir}:/scans/
      - ../../log/brotherscanner/:/var/log/
    ports:
      - 33355:33355
      - 54925:54925/udp # Scan Key Tool UDP: mandatory, for scanner tools
      - 54921:54921 # Scan Key Tool TCP: mandatory, for scanner tools
      - 161:161/udp # SNMP: mandatory, for scanner tools
    environment:
      - NAME=Paperless
      - MODEL=MFC-J4340DW
      - IPADDRESS=192.168.46.12
      - UID=500 # note: network mount needs to have correct permissions!
      - GID=101 # note: network mount needs to have correct permissions!
      - HOST_IPADDRESS=192.168.96.10
      - RESOLUTION=300
      - REMOVE_BLANK_THRESHOLD=0.55

      - WEBSERVER=true # optional, activates GUI & API
      - PORT=33355 # optional, sets port for webserver (default: 80)
      - DISABLE_GUI_SCANTOIMAGE=true
      - RENAME_GUI_SCANTOFILE="Scan front pages"
      - RENAME_GUI_SCANTOOCR="Scan front pages (B&W)"
      - RENAME_GUI_SCANTOEMAIL="Scan rear pages"
    restart: unless-stopped
```

pedropombeiro, 2024-10-11:
Looks like the problem is in `/etc/sudoers`, it lists the NAS user, instead of the user in `NAME`.

PhilippMundhenk, 2024-10-11:
I think that may be fixed in #32. I missed that when fixing username back then.

PhilippMundhenk, 2024-10-20:
closed in #58

PR #56 (2024-10-20): Fix api usage with custom name environment variable
Closes #55
closed 2024-10-20 by PhilippMundhenk