PhilippMundhenk/BrotherScannerDocker #61: Options greyed out on scanner

aacater, 2024-12-19:
I have a ADS-1700W. I have it connected to the server and it is working. The problem is, on the scanner, all the options are greyed out and can not be changed. Is there anyway to change the options?

This only happens when I use To Computer option. I can change options when I send to an FTP server.

PhilippMundhenk, 2025-02-16:
My apologies, I missed this issue. Not quite sure about this model. It is possible it does not use SNMP or uses a different set of options. The only way to find this out is to record the network traffic of the standard Brother tools with e.g., Wireshark. I can try to analyze this and compare. Maybe you can also provide a few pictures and more detailed description on how it is supposed to work and what exactly does not work. On older models, there is not greyed out options or possibility to change on the device...

PhilippMundhenk, 2026-10-10:
Thank you for the report, and apologies that it sat for so long. I went through all open issues today and wanted to give you a proper answer rather than more silence.

What you see is, as far as I can tell, how Brother's "Scan to PC" works by design rather than something this container does: when a scan is sent to a computer, the computer side owns the scan settings (resolution, colour mode, scan area), and the device panel shows them greyed out because they are not edited on the device. That is also why they become editable again for "Scan to FTP", where the device itself owns the job. The Brother Linux driver inside this container (`brscan-skey`) does not offer a way to hand those choices back to the panel, so there is nothing I can change here to make them editable on the ADS-1700W.

What has changed since you opened this: the settings can now be chosen on the container side instead. `RESOLUTION`, the new `SCAN_MODE` (for example `True Gray`, `Black & White`, as listed in the device capabilities at container start) and the scan window `SCAN_LEFT_MM`, `SCAN_TOP_MM`, `SCAN_WIDTH_MM`, `SCAN_HEIGHT_MM` are all environment variables now (see the README). If you scan in a few fixed ways, one container per profile with its own `NAME` shows up as separate destinations on the device, which gets close to choosing settings on the panel.

I am closing this for now since I cannot offer more without the device. If the container-side settings do not cover what you need, or if the ADS-1700W behaves differently from what I described, please reopen and attach the capabilities block from the container log (the `scanimage -A` output printed at startup) and a photo of the panel, and I will gladly look again. Thanks for using the project!

closed 2026-10-10 by PhilippMundhenk