PhilippMundhenk/BrotherScannerDocker #63: privileged is required for the docker image

grapemix, 2025-01-21:
In case someone face my problem, make sure privileged is set to true or this docker image won't work.

BTW, hosting a privileged docker image is not secure at all because a security breach from a single privileged image can lead to a host level security breach. Setting GID and UID won't help in this case. Docker image with non-root permission is always preferred.

PhilippMundhenk, 2025-01-21:
You do not require privileged mode for this container to work. The examples in the readme work fine.

What makes you think you require this? Anything that did not work?

grapemix, 2025-01-23:
I tried to NOT use privileged at first. I saw the log complaining unable to chown or chroot (I forget it). The log also shows a few part related to permission problem. Anyway, the server cannot start. After I switch to privileged mode, all errors are gone.

I did not dig into it since I am 95% sure I don't need to use this app at that time. I suspect granting some additional permission might be enough to run this app without privileged mode, but I never have time to test this theory.

PhilippMundhenk, 2025-02-26:
Tried to deploy this on a brand new WSL2 setup without privileged and with minimal configuration. Can't reproduce, scan is running fine.

It did lead me to discover that the UID & GID handling is not necessarily correct in some cases. Will tackle this in a separate issue.

Should this issue come up again, feel free to reopen.

closed 2025-02-26 by PhilippMundhenk