PhilippMundhenk/BrotherScannerDocker #64: Parsing error in lighttpd.conf

pedropombeiro, 2025-01-25:
Not sure if this is due to some recent update in the lighttpd package, but I'm seeing an error message when starting `ghcr.io/philippmundhenk/brotherscannerdocker:development`, and the web server is not accessible:

```
2025-01-25T09:50:37.988999907Z setting up webserver:
2025-01-25T09:50:37.989822361Z starting webserver for API & GUI...
2025-01-25T09:50:38.010051895Z running on port 33355
2025-01-25T09:50:38.380300058Z webserver started
2025-01-25T09:50:38.380336856Z -----
2025-01-25T09:50:38.380346671Z capabilities:
2025-01-25T09:50:38.380300962Z Duplicate config variable in conditional 0 global: url.rewrite-if-not-file
2025-01-25T09:50:38.390770946Z 2025-01-25 10:50:38: (configfile.c.2164) source: /etc/lighttpd/lighttpd.conf line: 70 pos: 0 parser failed somehow near here: (EOL)
```

pedropombeiro, 2025-01-25:
It seems that the problem stems from some repeated sections at the end of the `/etc/lighttpd/lighttpd.conf` config file:

```
#server.compat-module-load   = "disable"
server.modules += (
	"mod_dirlisting",
	"mod_staticfile",
)

server.modules += ( "mod_rewrite" )

url.rewrite-if-not-file = (
    "^/(.*)$" => "/index.php"
)

server.modules += ( "mod_rewrite" )

url.rewrite-if-not-file = (
    "^/(.*)$" => "/index.php"
)

server.modules += ( "mod_rewrite" )

url.rewrite-if-not-file = (
    "^/(.*)$" => "/index.php"
)
```

Once I removed them, I could start the server.

PhilippMundhenk, 2025-02-16:
fixed in #65

closed 2025-02-16 by PhilippMundhenk