PhilippMundhenk/BrotherScannerDocker #66: download.php error

peliopoulos, 2025-02-10:
I love the work you have done on this - with my work's painful admin rules, I can't really install the drivers natively, so I set this up on my home system and am able to run the scans.  Thanks!

The last hurdle is being able to download the files -
Steps to Reproduce:
1. I scan a document to file
2. Click on the Acrobat icon in the top corner
3. Click on a file name

Expected Result:
I expect the file to either open in the browser or download to the client computer

Actual Result:
This page isn’t working
192.168.50.150 is currently unable to handle this request.
HTTP ERROR 500

URL is http://localhost:8007/download.php?file=2025-02-09-18-27-00.pdf

Docker compose is:

services:
  brother-scanner:
    image: ghcr.io/philippmundhenk/brotherscannerdocker:v1.1.0
    container_name: brother-scanner
    volumes:
      - ./scans:/scans
    ports:
      - 8007:8007
      - 54925:54925/udp # mandatory, for scanner tools
      - 54921:54921 # mandatory, for scanner tools
      - 161:161/udp # mandatory, for scanner tools
    environment:
      - UID=1000
      - GID=1000
      - TZ=America/Vancouver            
      - NAME=Scanner
      - MODEL=DCP-L2540DW
      - IPADDRESS=192.168.1.125
      - HOST_IPADDRESS=192.168.1.150
      # - OCR_SERVER=localhost # optional, for OCR
      # - OCR_PORT=32800 # optional, for OCR
      # - OCR_PATH=ocr.php # optional, for OCR
      - WEBSERVER=true # optional, activates GUI & API
      - PORT=8007 # optional, sets port for webserver (default: 80)
      - DISABLE_GUI_SCANTOIMAGE=true # optional, deactivates button "Scan to image"
      - DISABLE_GUI_SCANTOOCR=true # optional, deactivates button "Scan to OCR"
      # - RENAME_GUI_SCANTOFILE="Scan front pages" # optional, renames button "Scan to file" to "Scan front pages"
      # - RENAME_GUI_SCANTOEMAIL="Scan rear pages" # optional, renames button "Scan to email" to "Scan rear pages"
      - REMOVE_BLANK_THRESHOLD=0.3
    restart: unless-stopped

PhilippMundhenk, 2025-02-11:
Indeed, I can reproduce that. Maybe @vgarcia007 can take a look.

closed 2025-02-16 by pedropombeiro

PhilippMundhenk, 2025-02-16:
@peliopoulos You may now use either tags latest or v1.1.1 and this should work.

peliopoulos, 2025-02-16:
thanks @PhilippMundhenk! works great now