PhilippMundhenk/BrotherScannerDocker #72: Instabilities

PhilippMundhenk, 2025-03-01:
Sometimes, after scanning (with OCR), an additional scan is not possible form the scanner itself. The situation usually recovers after a few minutes and scans are possible again. I can't reliably reproduce this but this seems to have started relatively recently in the past months ( fyi @pedropombeiro )

pedropombeiro, 2025-03-13:
I've also noticed that most of the time when I go to scan from the scanner control panel itself, it says no computer was found, but I can scan from the web page. I still haven't found the problem, but I can confirm that restarting the container fixes it for a while. I tested both `master` and `development`tags with the same result.

PhilippMundhenk, 2026-10-10:
Should be fixed in #90. Will reopen if not

closed 2026-10-10 by PhilippMundhenk

jhofker, 2026-10-10:
Fantastic!