PhilippMundhenk/BrotherScannerDocker #75: MDS-940DW - Partial page scan only

gitentropy, 2025-03-26:
Hi,

Firstly, thanks for the awesome image to utilise, it's a really convenient way to bring documents straight into my digital filing cabinet.

I do have an issue that has stumped me, and I was hopeful I've just missed something.

### Quick summary: 

The scans only capture the top ~1/4 of the page, the remainder is blank (pure white).

### Detailed:

I have a portable Brother Scanner MDS-940DW, while using the native app via my mobile/desktop/sd card, I am able to scan complete pages.

What I've tested:
1. Increasing the resolution 300 > 600 > 1200. But even at the higher end I only get ~3/4 of the page.
- This is also the upper limit I can apply before the it forcible rounds down.

Few problems with this 'solution':
- Scan still doesn't capture the full page as expected
- The physical scans take 4 x longer due to the resolution
- File size can be 20MB+

2. Change the blank pages from 0.0 to 1.0, incrementally by 0.1
- Didn't notice any change

3. Change the x/y parameters to match the document exactly
- This doesn't appear to have any affect on the outcome
- More often the values were rounded to larger values
- I've now defaulted it to the max allowed

### Setup

- Brother MDS-940DW connected via wifi
- Docker running under Debian 12
- Image: ghcr.io/philippmundhenk/brotherscannerdocker:v1.1.1

```
enviroment:
            - WEBSERVER=true
            - DISABLE_GUI_SCANTOIMAGE=true
            - DISABLE_GUI_SCANTOOCR=true
            - RENAME_GUI_SCANTOFILE="Start new scan"
            - RENAME_GUI_SCANTOEMAIL="Scan additional pages"
            - REMOVE_BLANK_THRESHOLD=0.3
```
### Log after Blank Threshold 0.3 and Resolution 300.

```
setting up user & logfile:
Allowing use of questionable username.
Adding user `Scanner' ...
Adding new user `Scanner' (1000) with group `NAS' ...
Creating home directory `/home/Scanner' ...
Copying files from `/etc/skel' ...
-----
setting up interface:
using interface: eth0
-----
setting up host IP:
-----
whole config:
=
IMAGE="bash  /opt/brother/scanner/brscan-skey/script/scantoimage.sh"
OCR="bash  /opt/brother/scanner/brscan-skey/script/scantoocr.sh"
EMAIL="bash  /opt/brother/scanner/brscan-skey/script/scantoemail.sh"
FILE="bash  /opt/brother/scanner/brscan-skey/script/scantofile.sh"
SEMID=b
eth=eth0
ip_address=***
-----
starting scanner drivers...
-----
setting up webserver:
starting webserver for API & GUI...
running on port 33355
webserver started
-----
capabilities:
Output format is not set, using pnm as a default.
Created directory: /var/lib/snmp/cert_indexes
scanimage: open of device brother4:net1;dev0 failed: Invalid argument
startup successful
/opt/brother/scanner/brscan-skey/script/scantofile.sh: line 7: UID: readonly variable
filename: /tmp/2025-03-26-22-48-33/2025-03-26-22-48-33-front-page%04d.pnm
scanimage: rounded value of br-x from 211.881 to 211.881
scanimage: rounded value of br-y from 355.567 to 355.567
Scanning infinity pages, incrementing by 1, numbering from 1
Scanning page 1
Scanned page 1. (scanner status = 5)
Scanning page 2
Scanned page 2. (scanner status = 5)
Scanning page 3
scanimage: sane_start: Document feeder out of documents
Batch terminated, 2 pages scanned
conversion process for 2025-03-26-22-48-33 is running in PID: 1297
converting to PDF for 2025-03-26-22-48-33...
SSH environment variables not set, skipping inotify trigger.
TELEGRAM_TOKEN or TELEGRAM_CHATID environment variables not set, skipping Telegram trigger.
FTP environment variables not set, skipping inotify trigger.
cleaning up for 2025-03-26-22-48-33...
OCR environment variables not set, skipping OCR.
```

PhilippMundhenk, 2026-02-08:
Is this issue really from March 2025? My apologies! I must have totally missed it!!

Not quite sure what is wrong here. The logs show a correct scan of two pages. Could it be connectivity issues between scanner and server? An instable WiFi connection or something? This is basically all handled in the scanner driver by Brother, but I'm only aware of two cases, where this has happened: Either loss of connectivity, or scanner reports end of page (either because really finished, or because stop button pressed). 
What stumps me here is that the resolution makes a difference.
I think @yashdavisgupta is referring to the settings of the scanner itself, not the web GUI of this project. Not sure which option, though.

Is the page actually fully scanned but doesn't show up? Or is it scanned 1/4 only and then quickly run through for the rest of the way (as if you press stop after 1/4)?

yashdavisgupta, 2026-02-08:
Looking back, updating the firmware to the latest version and resetting the device fixed it for me.

PhilippMundhenk, 2026-10-10:
Since #123 the scan window is configurable without editing the scripts: `SCAN_HEIGHT_MM`, `SCAN_WIDTH_MM`, `SCAN_TOP_MM`, `SCAN_LEFT_MM`, plus `SCAN_MODE` and `RESOLUTION`. If the partial page persists with the current image, the full `scanner.log` of one scan would help; it now logs the exact `scanimage` command that was run.

gitentropy, 2026-10-13:
[_scanner_logs.txt](https://github.com/user-attachments/files/33378217/_scanner_logs.txt)

Thanks for the updated features.

I've done some additional testing after updating, ultimately getting the same results as previous.

Attempted a few different combinations, some failing all together - using over 1200dpi for example.

Interestingly, the missing portion is relative to the page size. So scanning an A4 sheet misses the bottom 1/5, and same will occur if scanning an A5.

PhilippMundhenk, 2026-10-13:
Thanks for the log, that was the missing piece. Your DS-940DW is not in the `brscan4` driver's model tables: `brsaneconfig4` accepts the name but registers the device as "Unknown", and the backend then scans with a generic profile. That is a likely reason for the proportional cut-off and for the I/O errors above 1200 dpi (the generic profile offers interpolated resolutions the device does not have). Brother's own DS-940DW FAQs point to the `brscan5` driver, and `brsaneconfig5` does recognise the model.

The current master image installs both drivers and picks brscan5 for your model automatically (#173, #176). Please pull `ghcr.io/philippmundhenk/brotherscannerdocker:latest`, keep your settings, and check the start of the log for `scanner driver: brscan5 for model DS-940DW` and the capability list the device reports now. A test scan at 300 dpi would tell whether the page is complete; if not, the full log of that scan again, please.

gitentropy, 2026-10-13:
Fantastic, that has worked. 

Page now fully scanned at 300dpi.

Tested with A4 and A5. A5 scans the full A4 size, leaving dead space on the side and bottom, though I haven't changed any threshold settings from my previous testing - and isn't really a concern for me.

```
-----
starting scanner drivers...

scanner driver: brscan5 for model MDS-940DW
-----
```

I believe this closes the issue off, really appreciate the efforts.

closed 2026-10-13 by gitentropy